kata0510/Lily58 #1: Kailhロープロファイルだとアクリルケース邪魔でキーキャップが設置出来ない

t10471, 2018-10-27:
Kailhロープロファイル対応と書いてあると思うのですが、
遊舎工房さんで買った `Kailhロープロファイルスイッチ` と `Kailhロープロ無刻印(遊刻印)キーキャップ` だとキーの高さに対して上のアクリルケースが邪魔でキーキャップが設置出来ません。
Kailhロープロファイル用のアクリルケースとスペーサーが欲しいです。

t10471, 2018-10-27:
これがはめた画像です。
[image]

遊舎工房さんで(https://yushakobo.jp/shop/pg1350/)買ったものだと、
アクリルケースの穴よりちょっと大きい気がします...
[image]

kata0510, 2018-10-27:
写真を見させていただいたところ組み立て方に間違いがあるようです。
ロープロファイルの場合PCBとスイッチをマウントするアクリルの間に隙間が生まれないためダイオードは**裏側に**はんだ付けをする必要があります。
スイッチをアクリルに取り付けるにはコツがありますが斜めの状態でスイッチ側面の爪の部分を押し込むと取り付けられるかと思います。
この場合ですと一旦スイッチのはんだ付けをすべて外し、ダイオードが裏側に来るように左右のPCBを逆にしてからスイッチをアクリルに取り付けてサイドはんだ付けを行う必要があります。

t10471, 2018-10-28:
なるほど、、、もう一セット買おうかと思います。。。

closed 2018-10-28 by t10471

kata0510, 2018-10-28:
半田吸い取り線でスイッチを取り外して正しく組み直せば十分対処できる範囲かと思います。また何か質問などあればTwitterにて対応出来ますのでよろしくお願いします。