kata0510/Lily58 #3: Problems with the gerber files (non pro lily58)

caileighf, 2019-04-26:
Hi! I tried to order the PCBs from a fab shop I've used before but it looks like the gerber only has the copper traces. Would you be willing to upload gerbers with all the necessary layers? Or if you could let me know how it's supposed to be done in KiCad if that's were you designed it. I really love the board you've made.
[image]

kata0510, 2019-05-06:
Please plot the file with the setting suitable for the PCB factory by the procedure of File-> Plot
The uploading Gerber is for elecrow
thanks!
[image]
[image]
[image]

caileighf, 2019-05-07:
This is great! Thank you for the instructions.

closed 2019-05-07 by caileighf