kata0510/Lily58 #8: Right half not working

BenRoe, 2019-10-14:
Hello,
i need some help.
I flashed the pro micro's with the default rev1 keymap. `make lily58/rev1:default:avrdude`.  

If i test both side separate (via USB), they send keycodes. The right half has the layout from the left half.

If i connect the left half via USB and connect both sides with the trrs cabel (4pole), left works, but right sends no keycode.
Any idea what the problem could be?

Disabled the backlight in the rule.mk, desoldered the bridges for the oled.

[image]

BenRoe, 2019-10-15:
I also soldered mx and choc sockets. This shouldn't be a problem, right?

BenRoe, 2019-10-15:
Ok i fixed the problem.
Seems that some pins are not connected, because i used the [spring pin header](https://user-images.githubusercontent.com/6285554/57210525-f5171480-7017-11e9-9d92-3a345d53db94.jpg).

Now i use the mill-max pin header and everything works.
Hope this helps, if someone has a similar problem to fix it faster.

closed 2019-10-15 by BenRoe

ikakus, 2019-12-26:
Hi! I have the same problem.
I have a hoy swap sockets for pro-micros. I've tried 3 different pro-micro combinations still the problem remains. 

My guess is, maybe I've damaged this conduction "road". Can you please advise if I can connect the damaged circuit with a wire directly from pro micro pin to TRRS jack? If yes, which one should I connect?

[image]