kata0510/Lily58 #10: Right half not working (Lily58 Pro)

ikakus, 2019-12-26:
Hi! I have a problem.
I flashed the pro micro's with the default rev1 keymap. make lily58/rev1:default:avrdude.

If I test both side separately (via USB), they send keycodes. The right half has the layout from the left half.

If I connect the left half via USB and connect both sides with the trrs cable (4pole), left works, but right sends no keycode.
Any idea what the problem could be?
I have hot swap sockets for pro-micros. I've tried 3 different pro-micro combinations still the problem remains. 

My guess is, maybe I've damaged this conduction "road". Can you please advise if I can connect the damaged circuit with a wire directly from pro micro pin to TRRS jack? If yes, which one should I connect?

[image]

_Originally posted by @ikakus in https://github.com/kata0510/Lily58/issues/8#issuecomment-569059133_

kata0510, 2019-12-26:
Hi
Using blue ProMicro?
Blue ProMicro recommends using black ProMicro due to left / right communication issues.

ikakus, 2019-12-26:
Yes, blue ones. Oh wow, I thought they are all the same!
Thanks, I will try with black ones!
[image]

ikakus, 2019-12-26:
I think I've found the problem that you mentioned, but it is about **green** ones.
[green micro vs blue](http://www.40percent.club/2017/09/green-pro-micro.html).
Altho the only black micros that I've found are Elite-C, were you referring to them?

BenRoe, 2019-12-26:
I have the same problem with a Lily58 keyboard from a customer. Tomorrow i replace the pro micro with different pro micro's and elite-c and post the result here.

There is another fix. A different slave detection method ([PR](https://github.com/qmk/qmk_firmware/pull/6424)).

> “A few of us collaborators of the QMK Firmware reviewed and tested the new option, signed off on it, and then it got merged in. So to use the new detection method (which does not rely on the Schottky diode), in config.h of your keymap, add the…” — @keebio https://link.medium.com/kmZ3dambK2

Would be great if the Lily58 uses the new `split_common` framework.

kata0510, 2019-12-27:
The problem that blue ProMicro cannot communicate left and right has been reported in Japan.
https://twitter.com/yohewi/status/1204048124244717568

I can't verify because I don't have Elite-C. sorry.

ikakus, 2019-12-27:
@BenRoe, I'm using regular pro micros (with micro-usb port), do you think this is still related?

BenRoe, 2019-12-27:
Did some test with different pro micro's. 
Left one works if on both sides. Also the Elite-C works. Will compare both under a magnifier to find difference. 
First obvious difference is the micro usb port.
[image]

CurryFurious, 2020-01-02:
YMMV, But for me I switched to a from a 4 pole TRRS cable to a 3 pole TRS Cable, and for whatever reason it actually work. If you want to try the lily58 with split common code, check out my personal fork of qmk, it has both the CRKBD and Lily58 working with split common. You might need to add `SPLIT_TRANSPORT = yes` in your keymaps rules.mk for it to work properly, or you can try my keymap (make lily58:curry:flash), however it is meant for the elite-c.

ikakus, 2020-01-04:
@CurryFurious I've tried with 3 pole TRS, still same issue. I'm still waiting for black ProMicro's to arrive, will post an update once I'll try them.

ikakus, 2020-01-15:
So, I've finally got to try **black** pro-micros on both sides, and it has the same issue. Can you please tell me what pins are used for two sides connection I'll try to solder them directly.

[image]

aWalrus, 2020-01-17:
I had a similar issue! Are you, by any chance, using RGB LED backlight?
6 LEDs per half in my build (with full brighness) was too much and lead to brownouts - with one half not working.
As soon as I disabled them, the issue was gone.

ikakus, 2020-01-17:
@aWalrus Hi! No, I have only both types of sockets, reset switch, trrs and pro-micros soldered.

ikakus, 2020-01-18:
@kata0510 can I use this connection? I want to solder them directly to exclude pcb problem?
[image]

ikakus, 2020-01-20:
So I was able to fix the issue. It appeared physical damage as I suggested before. 
As SOFT_SERIAL_PIN set to PD2 in file serial_config.h. I've just connected them directly with a wire. 
[image] and it solved the problem.

Then I just soldered PD2 pin to TRRS on the damaged board.
[image]

No problems so far :)

closed 2020-01-20 by ikakus

BiotechPedro, 2021-02-28:
> YMMV, But for me I switched to a from a 4 pole TRRS cable to a 3 pole TRS Cable, and for whatever reason it actually work. If you want to try the lily58 with split common code, check out my personal [...]

@Curry I've tried with your keymap but still I suffer the same problem... using 2 elite-c and trying to flash a lily58 default keymap, the right part doesn't do anything. Sometimes the left hand functions perfectly and others only "z", "x", "b", "v" "[" and " " works. Can you help me please?

fi0, 2021-06-07:
@BiotechPedro which version of elite-c are you using?

hzx, 2023-01-05:
in keymap/config.h you need to define:
#define SPLIT_USB_DETECT
also choose hand for each promicro:
controller have left hand buttons
#define MASTER_LEFT
controller have right hand buttons
#define MASTER_RIGHT

NotNite, 2023-04-17:
Running into this too - I tried with SPLIT_USB_DETECT but nothing.