kata0510/Lily58 #13: PCB outline is not closed

B-Raffai, 2020-05-10:
[image]
The outline of the PCB is open preventing step file export and breaking the 3D viewer.

If you manually adjust the outline to make the 3D viewer work, the step export will still break with the following messages. 

```
Executing ‘"/usr/bin/kicad2step" --drill-origin -f -o "/home/me/source/Lily58/pcb/Lily58.step" "/home/me/source/Lily58/pcb/Lily58.kicad_pcb"’
Warning: 09:10:10: /build/kicad/src/kicad-5.1.5/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'J2'
Warning: 09:10:10: /build/kicad/src/kicad-5.1.5/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'R1'
Warning: 09:10:10: /build/kicad/src/kicad-5.1.5/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'R1'
Warning: 09:10:10: /build/kicad/src/kicad-5.1.5/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'R2'
Warning: 09:10:10: /build/kicad/src/kicad-5.1.5/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'R2'
Warning: 09:10:10: /build/kicad/src/kicad-5.1.5/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'P2'
Warning: 09:10:10: /build/kicad/src/kicad-5.1.5/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'P2'
Warning: 09:10:10: /build/kicad/src/kicad-5.1.5/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'P1'
Warning: 09:10:10: /build/kicad/src/kicad-5.1.5/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'P1'
Warning: 09:10:10: /build/kicad/src/kicad-5.1.5/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'W1'
Warning: 09:10:10: /build/kicad/src/kicad-5.1.5/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'JP1'
Warning: 09:10:10: /build/kicad/src/kicad-5.1.5/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'JP2'
Warning: 09:10:10: /build/kicad/src/kicad-5.1.5/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'JP3'
Warning: 09:10:10: /build/kicad/src/kicad-5.1.5/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'JP4'
Warning: 09:10:10: /build/kicad/src/kicad-5.1.5/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'JP5'
Warning: 09:10:10: /build/kicad/src/kicad-5.1.5/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'JP6'
Warning: 09:10:10: /build/kicad/src/kicad-5.1.5/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'JP7'
Warning: 09:10:10: /build/kicad/src/kicad-5.1.5/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'JP8'
Warning: 09:10:10: /build/kicad/src/kicad-5.1.5/utils/kicad2step/pcb/oce_utils.cpp: CreatePCB: 754
Warning: * could not close outline (dropping outline data with 25 segments)
Warning: + line start: 6.99333,-82.9895 end: 5.29176,-79.3181
Warning: + line start: 5.29176,-79.3181 end: 3.86311,-75.5307
Warning: + line start: 3.86311,-75.5307 end: 2.688,-71.6491
Warning: + line start: 2.688,-71.6491 end: 1.74708,-67.6952
Warning: + line start: 1.74708,-67.6952 end: 1.02097,-63.6908
Warning: + line start: 1.02097,-63.6908 end: 0.490297,-59.6577
Warning: + line start: 0.490297,-59.6577 end: 0.13569,-55.6176
Warning: + line start: 0.13569,-55.6176 end: -0.062223,-51.5923
Warning: + line start: -0.062472,-51.5922 end: -0.062472,-45.2853
Warning: + line start: -0.062472,-45.2853 end: -0.062472,-38.9784
Warning: + line start: -0.062472,-38.9784 end: -0.062472,-32.6714
Warning: + line start: -0.062472,-32.6714 end: -0.062472,-26.3645
Warning: + line start: -0.062472,-26.3645 end: -0.062472,-20.0576
Warning: + line start: -0.062472,-20.0576 end: -0.062472,-13.7506
Warning: + line start: -0.062472,-13.7506 end: -0.062472,-7.44369
Warning: + line start: -0.062472,-7.44369 end: -0.062472,-1.13676
Warning: + line start: -0.06179,-1.13676 end: -0.040477,-0.925015
Warning: + line start: -0.040477,-0.925015 end: 0.020657,-0.727855
Warning: + line start: 0.020657,-0.727855 end: 0.117409,-0.549489
Warning: + line start: 0.117409,-0.549489 end: 0.24557,-0.394123
Warning: + line start: 0.24557,-0.394123 end: 0.400937,-0.265961
Warning: + line start: 0.400937,-0.265961 end: 0.579303,-0.16921
Warning: + line start: 0.579303,-0.16921 end: 0.776462,-0.108075
Warning: + line start: 0.776462,-0.108075 end: 0.98821,-0.086762
Warning: + line start: 0.98821,-0.086762 end: 11.396,-0.086762
[... 54 lines cut ...]
Warning: + line start: 95.3193,-93.3799 end: 95.3193,-94.598
Warning: + line start: 95.3193,-94.598 end: 95.3193,-95.816
Warning: + line start: 95.3193,-95.816 end: 95.3193,-97.0117
Warning: + line start: 95.3193,-97.0117 end: 95.2889,-97.3142
Warning: + line start: 95.2889,-97.3142 end: 95.2016,-97.5958
Warning: + line start: 95.2016,-97.5958 end: 95.0633,-97.8506
Warning: + line start: 95.0633,-97.8506 end: 94.8803,-98.0726
Warning: + line start: 94.8803,-98.0726 end: 94.6583,-98.2557
Warning: + line start: 94.6583,-98.2557 end: 94.4035,-98.3939
Warning: + line start: 94.4035,-98.3939 end: 94.1218,-98.4812
Warning: + line start: 94.1218,-98.4812 end: 93.8193,-98.5117
Warning: + line start: 93.8193,-98.5117 end: 86.4498,-98.5117
Warning: + line start: 86.4498,-98.5117 end: 79.0803,-98.5117
Warning: + line start: 79.0803,-98.5117 end: 71.7108,-98.5117
Warning: + line start: 71.7108,-98.5117 end: 64.3413,-98.5117
Warning: + line start: 64.3413,-98.5117 end: 56.9717,-98.5117
Warning: + line start: 56.9717,-98.5117 end: 49.6022,-98.5117
Warning: + line start: 49.6022,-98.5117 end: 42.2327,-98.5117
Warning: + line start: 42.2327,-98.5117 end: 34.8632,-98.5117
Warning: 09:10:10: /build/kicad/src/kicad-5.1.5/utils/kicad2step/pcb/oce_utils.cpp: CreatePCB: 754
Warning: * could not close outline (dropping outline data with 1 segments)
Warning: + line start: 21.8078,-0.086762 end: 11.4,-0.086762
Warning: 0x5629013fed00 : Standard_NullObject: A null Label has no attribute.
Error: Unable to create STEP file. Check that the board has a valid outline and models.
```

jpuerto96, 2020-09-11:
Any update on this? I am also running into the same issue when attempting to export this file to STEP.

pp222, 2020-09-17:
I just fixed right now. Will be uploading pcb file in a couple of mins.

pp222, 2020-09-17:
> Any update on this? I am also running into the same issue when attempting to export this file to STEP.

@jpuerto96 check my repository... there the pcb file is the updated one. Let me know if it's ok.

jpuerto96, 2020-09-23:
@chimuelo22 Doesn't seem like it is fixed in your repository.

`Executing '"/private/var/folders/bw/xt5q6ktx6mb1p1h21cdc12z80000gn/T/AppTranslocation/9CB082CB-75AD-4E1E-9BA5-2930222771CD/d/kicad.app/Contents/Applications/pcbnew.app/Contents/MacOS/../../../../MacOS/kicad2step" --drill-origin -f -o "/Users/juan/Lily58/pcb/Lily58.step" "/Users/juan/Lily58/pcb/Lily58.kicad_pcb"'
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'J2'
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'R1'
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'R1'
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'R2'
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'R2'
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'P2'
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'P2'
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'P1'
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'P1'
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'W1'
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'JP1'
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'JP2'
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'JP3'
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'JP4'
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'JP5'
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'JP6'
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'JP7'
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: AddComponent: 588
Warning: * no model defined for component 'JP8'
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: CreatePCB: 754
Warning: * could not close outline (dropping outline data with 25 segments)
Warning: + line start: 6.99333,-82.9895 end: 5.29176,-79.3181
Warning: + line start: 5.29176,-79.3181 end: 3.86311,-75.5307
Warning: + line start: 3.86311,-75.5307 end: 2.688,-71.6491
Warning: + line start: 2.688,-71.6491 end: 1.74708,-67.6952
Warning: + line start: 1.74708,-67.6952 end: 1.02097,-63.6908
Warning: + line start: 1.02097,-63.6908 end: 0.490297,-59.6577
Warning: + line start: 0.490297,-59.6577 end: 0.13569,-55.6176
Warning: + line start: 0.13569,-55.6176 end: -0.062223,-51.5923
Warning: + line start: -0.062472,-51.5922 end: -0.062472,-45.2853
Warning: + line start: -0.062472,-45.2853 end: -0.062472,-38.9784
Warning: + line start: -0.062472,-38.9784 end: -0.062472,-32.6714
Warning: + line start: -0.062472,-32.6714 end: -0.062472,-26.3645
Warning: + line start: -0.062472,-26.3645 end: -0.062472,-20.0576
Warning: + line start: -0.062472,-20.0576 end: -0.062472,-13.7506
Warning: + line start: -0.062472,-13.7506 end: -0.062472,-7.44369
Warning: + line start: -0.062472,-7.44369 end: -0.062472,-1.13676
Warning: + line start: -0.06179,-1.13676 end: -0.040477,-0.925015
Warning: + line start: -0.040477,-0.925015 end: 0.020657,-0.727855
Warning: + line start: 0.020657,-0.727855 end: 0.117409,-0.549489
Warning: + line start: 0.117409,-0.549489 end: 0.24557,-0.394123
Warning: + line start: 0.24557,-0.394123 end: 0.400937,-0.265961
Warning: + line start: 0.400937,-0.265961 end: 0.579303,-0.16921
Warning: + line start: 0.579303,-0.16921 end: 0.776462,-0.108075
Warning: + line start: 0.776462,-0.108075 end: 0.98821,-0.086762
Warning: + line start: 0.98821,-0.086762 end: 11.396,-0.086762
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: CreatePCB: 754
Warning: * could not close outline (dropping outline data with 71 segments)
Warning: + line start: 34.8632,-98.5117 end: 42.2327,-98.5117
Warning: + line start: 42.2327,-98.5117 end: 49.6022,-98.5117
[... 48 lines cut ...]
Warning: + line start: 142.46,-63.3265 end: 142.46,-53.8449
Warning: + line start: 142.46,-53.8449 end: 142.46,-44.3632
Warning: + line start: 142.46,-44.3632 end: 142.46,-34.8816
[... cut ...]
Warning: + line start: 53.0274,-0.086762 end: 42.6195,-0.086762
Warning: + line start: 42.6195,-0.086762 end: 32.2117,-0.086762
Warning: + line start: 32.2117,-0.086762 end: 21.8039,-0.086762
Warning: 10:35:07 AM: /vagrant/build/kicad/src/kicad/utils/kicad2step/pcb/oce_utils.cpp: CreatePCB: 754
Warning: * could not close outline (dropping outline data with 1 segments)
Warning: + line start: 21.8078,-0.086762 end: 11.4,-0.086762
Error: Unable to create STEP file. Check that the board has a valid outline and models.`

kata0510, 2020-09-23:
outline has been fixed.
Please check.

closed 2020-09-23 by kata0510

jpuerto96, 2020-09-23:
@kata0510 I just pulled from master and did the following:

- Open KiCAD
- Open file (Lily58/pcb/Lily58.kicad_pcb) in KiCAD
- File>Export>STEP
- [image]

I may be doing something wrong, but it doesn't seem to export, I get the output previously described when attempting the export.

kata0510, 2020-09-23:
@jpuerto96 
sorry.
We don't support exporting STEP files until the STEP file is exported.
I can't make any changes as I'm manufacturing & selling with this data.

You may be able to solve this problem by drawing a new outline.

jpuerto96, 2020-09-23:
@kata0510 Understood - thank you for the clarification.

DanBlack3232, 2023-04-03:
HI
I'm not familiar with pcb program and i just downloaded kicad latest version but when i try to open Lily58_Pro.kicad_pcb file i get this error :
[image]
[image]
and its not just this file , i can't open any other projects.
can any one help me?
Thanks