kata0510/Lily58 #19: JLCPCB cannot produce the Pro Top file

fireglow, 2020-10-19:
I've created a gerber file, using [this guide](https://support.jlcpcb.com/article/44-how-to-export-kicad-pcb-to-gerber-files), of the Lily58 Pro top plate ([attached](https://fireglow.de/755/gerberTOP.zip)) and sent it to JLCPCB for production, but they say:
>There are too many small slots in your file , it will take too much time to get it finished during the routing process.

How can I remedy this?

Seems like I can't upload the gerber zip file to GitHub, I've uploaded it here: https://fireglow.de/755/gerberTOP.zip

fireglow, 2020-10-19:
Oh, I just saw, they want another €12.30 to make it.
Black LeadFree HASL, 10 pieces, their initial estimate was €12.27.

jpconstantineau, 2020-10-19:
A few months ago, they have started asking more for keyboard plates for this reason.  They are also more picky when inspecting the gerbers and have been rejecting more.

kata0510, 2020-10-19:
Use a different PCB manufacturer. I use elecrow, ALLPCB.

fireglow, 2020-10-20:
Okay I see, this is not a problem with the CAD files, very good.
Thank you for the recommendations, I was looking for viable alternatives for a while now, will use them when I need something printed again.

closed 2020-10-20 by fireglow