kata0510/Lily58 #21: bad pcb tracing

heartpolik, 2020-11-20:
have an gap at this place
http://joxi.ru/823vQjec8aBzNm
http://joxi.net/zAN6l5ZTj8qw8A

kata0510, 2020-11-30:
Open in Kicad and there is no problem
[image]
[image]

closed 2021-03-04 by kata0510

japko, 2024-06-03:
I just want to leave a comment for people having same issue.

[image]

Debugging this was a nightmare. I'm not so skilled in soldering so I assumed this was my fault ;-)

I ordered PCB at jlcpcb.com