kata0510/Lily58 #23: Feature req: comparison matrix

dimaqq, 2021-02-16:
It would be great to see what the difference between `pro` and `lite v2` is at a glance.
(as far as I can see these are the two generally available options now).

kata0510, 2021-03-04:
The difference in appearance between Pro and Lite Rev2 is the presence or absence of the i2c port.
The Pro uses a socket to mount the switch.
Lite Rev2 has the switch soldered in place

Here is a page with a brief comparison (Japanese)
[https://kata0510.github.io/Lily58-Document/start.html](https://kata0510.github.io/Lily58-Document/start.html)

dimaqq, 2021-03-04:
Maybe it's worth it to add this top-level readme :)

mmai, 2021-04-26:
What do the triangles and circles mean on the page?