kata0510/Lily58 #34: Updated PCB with controller integrated, hot swap and RGB per key

kevinduffey, 2022-01-20:
Been a big fan of Keeb.io Iris keyboard. Only things I desired for it were per key RGB, and I wish there was a single thumb key and single index key.. which the Lily58 design seems to have.

But.. Iris as of rev4 integrated the controller into the PCB so no more ugly controller chip sticking out and easy to damage. The new Rev6 (just released) now adds hot swap and per key RGB integrated in to the PCB. Literally no longer need to do any soldering.. which for some may be an issue.. but for many of us we just want a great keyboard with the ability to choose switches and caps. 

I would like to know if there is any chance an updated Lily58 PCB can be created that does what Iris Rev 6 did.. integrate the controllers on each half, the per key RGB and the MX hot swap sockets.

With that, I think I personally would have my absolute favorite keyboard. Love my Iris.. but miss the Model 01 tab/enter keys for index fingers.. which Lily58 provides. Please make it happen. I'd pay $80 for that PCB which is what the Iris Rev6 with everything in it sells for.

BenRoe, 2022-01-22:
There is already Lily58 version available what you want, except the onboard controller. It’s called Lily58L

kevinduffey, 2022-09-11:
Yup.. I discovered this. I was hoping given that Boardsource.xyz built the Lulu from the PCB (lily58L I believe).. and how nice that came out.. that there would be a ready to use PCB with everything soldered on.. so that you just buy the pcb, case, switches, and assemble. The IRIS Rev 6 for example is $70 for the fully integrated PCB with per key RGB, hot swap switches and back light RGB LEDs and the controller, USBC between the halves, etc. I was hoping there was a chance this could be done for the Lily58 as well.

kata0510, 2023-01-08:
Lily58 Pro Rev2 is under development. Backlight LED and Type-C are supported (planned).

Serhi0, 2023-01-22:
> Lily58 Pro Rev2 está en desarrollo. Se encontró retroiluminación LED y tipo C (planificado).

when do you think you could release rev 2?

winterdouglas, 2023-12-13:
Any updates about this? :)