kata0510/Lily58 #35: Lifted the oled jumper pads, is there a workaround?

thraizz, 2022-03-07:
Hi, this was my first soldering project and i think i destroyed one of the oled jumper pads. The right split's oled now stays dark when connecting the keyboard. Can i somehow workaround this? The pads on the other side are fine, can I just use them? Thanks for any help in advance :)

twinoner, 2022-03-07:
using the other side pads won't help you, because that's for the other side.
In any case, and with a bit of luck and patience, you can open a little bit of the burned pads and join them via a small cable or something like that.

thraizz, 2022-03-07:
Thanks for your advice @twinoner , just using the other side is maybe a good idea - I noticed that i also soldered all leds on the wrong side of the board. I think i will desolder the pro micros, reset buttons and TRRS sockets and flip the pcbs :cat:

twinoner, 2022-03-07:
go nuts, man.. but remember to also desolder the diodes.. it will work

thraizz, 2022-03-07:
They are currently on the same side as the promicros and switches, so they can stay there, right? :D

twinoner, 2022-03-07:
I figured.. then go nuts, flip the promicros and the sockets.. that way you don't have to buy another board.. 
wait.. what board you have? because the lily58/pro doesn't have RGB.. or is it a custom one? I'm interested..

thraizz, 2022-03-07:
Its the lily58 rev1,can be bought as a full kit here: https://42keebs.eu/shop/kits/lily58-split-ergo-50-kit-black/

twinoner, 2022-03-07:
though it was the hot swappable.. then go nuts..
unsolder the switches, diodes (if needed) and the micro, flip them and you should be good to go..