kata0510/Lily58 #59: Help Identifying PCB

snot-x, 2025-09-03:
I am sorry if this is not the appropriate method of asking a question, but I was given a set of old lily58 pcbs and I am unable to identify exactly which model they are as they are missing almost any identifying marks aside from the name lily58.
I have went ahead and ordered 1n4148 diodes assuming that those will be correct, but the thing I am wondering is what direction to solder them in, as the documentation shows an arrow that is missing from my PCBs.

[image]

twinoner, 2025-09-03:
I'm not sure, but this seems to be a "normal" Lily58 (non PRO version, the PRO is the one with hotswap)
Also, this apparently is a 'one-sided' board (so you have the LEFT and RIGHT ones)
hope this helps https://www.reddit.com/r/MechanicalKeyboards/comments/9qpfta/comment/e8aqn4v/
in the bottom left is the name of the designer of the board

twinoner, 2025-09-04:
@snot-x you can always just solder two diodes (one on each side) and test it that way
or, and I'm not a pro with tester, try it with a tester for continuity or something like that 🤔